arduino/arduino-examples #47: 01.Basics > Blink does not compile for Arduino Nicla Sense ME

gmacario, 2021-10-03:
Hello,

I followed the documentation published on link "Built-in Examples" of <https://docs.arduino.cc/hardware/nicla-sense-me> and attempted to build example "01.Basics > Blink" to the Arduino Nicla Sense ME.

However the compilation fails with the following error:

TODO

Any suggestions?

P.S.: My apologies if this is not the proper GitHub repository to report the issue. Please feel free to move elsewhere.

per1234, 2021-10-03:
Hi @gmacario. Thanks for your report. The Nicla has its own equivalent to the Blink example. The standard Blink example you find in this repository can't be used with that board.

There are instructions for it here:
https://docs.arduino.cc/software/ide-v1/tutorials/getting-started/cores/arduino-mbed_nicla#uploading-a-simple-example

Enjoy using your new Nicla Sense ME!

closed 2021-10-03 by per1234

gmacario, 2021-10-03:
> Hi @gmacario. Thanks for your report. The Nicla has its own equivalent to the Blink example. The standard Blink example you find in this repository can't be used with that board.

I see... Thank you very much for your hint!