arduino/arduino-examples #63: [AE-60] `Blink.ino` does not compile for Nicla Sense ME, unless `#include "Nicla_System"`

aliphys, 2023-04-27:
### Describe the problem

The Blink sketch does not run without modification on the Nicla Sense ME. It is required to add the `#include "Nicla_System.h"` directive, in order for the sketch to compile and successfully upload.

### To reproduce

1. Open [Blink.ino](https://github.com/arduino/arduino-examples/blob/eb709fa68aca89641f4ff4da299cc63244749e6d/examples/01.Basics/Blink/Blink.ino) from **File -> Examples ->  01.Basics -> Blink**.
2. Connect the Nicla Sense ME board and select from the drop down menu
3. Compile. A message informing the user to include the Nicla_System.h library is given in the bottom right corner
[image]
```
/tmp/.arduinoIDE-unsaved2023327-15341-54bj3y.wzmz7/Blink/Blink.ino: In function 'void setup()':
/tmp/.arduinoIDE-unsaved2023327-15341-54bj3y.wzmz7/Blink/Blink.ino:29:10: error: call to 'pinMode' declared with attribute error: Please include Nicla_System.h to use this pin
   pinMode(LED_BUILTIN, OUTPUT);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved2023327-15341-54bj3y.wzmz7/Blink/Blink.ino: In function 'void loop()':
/tmp/.arduinoIDE-unsaved2023327-15341-54bj3y.wzmz7/Blink/Blink.ino:34:15: error: call to 'digitalWrite' declared with attribute error: Please include Nicla_System.h to use this pin
   digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
   ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved2023327-15341-54bj3y.wzmz7/Blink/Blink.ino:36:15: error: call to 'digitalWrite' declared with attribute error: Please include Nicla_System.h to use this pin
   digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
   ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: call to 'pinMode' declared with attribute error: Please include Nicla_System.h to use this pin
```
4. Add `#include "Nicla_System.h"` just above the `setup()` function.
5. Compile again, which will be successful.
```
Sketch uses 55788 bytes (10%) of program storage space. Maximum is 527616 bytes.
Global variables use 12256 bytes (19%) of dynamic memory, leaving 52032 bytes for local variables. Maximum is 64288 bytes.
```
7. After uploading, all three RGB LEDs will start blinking together (white light)

### Expected behavior

It is NOT required to add `#include "Nicla_System.h" to the sketch. Based on the selected FQBN, the compiler should add the directive if required. 

### The examples version

https://github.com/arduino/arduino-examples/commit/dfedf56610d6110887e227e114c11c2ba4211fb7

### Additional context

The Blink sketch is the de-facto Hello World example, and often **the first sketch users upload** to their board.
Replicated on IDE 1 and Web Editor also.
[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-examples/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

aliphys, 2023-04-27:
While there is a dedicated Blink_Nicla.ino sketch, the Blink.ino sketch should work with all boards.  
- https://github.com/arduino/ArduinoCore-mbed/issues/615

facchinm, 2023-04-28:
The `error` directive was added to target EXACTLY that case; I think it's more than enough to drive the user to add the include (which is enough to make everything work automatically)

closed 2024-05-22 by per1234