arduino/arduino-examples #83: Starter Kit Digital Hourglass out-of-bounds error

drf5n, 2025-02-15:
https://github.com/arduino/arduino-examples/blob/2c2bb7bc82fe4a24fbd81ff8f5fa6004e5cf2e86/examples/10.StarterKit_BasicKit/p08_DigitalHourglass/p08_DigitalHourglass.ino#L57

Note that the code in the Digital Hourglass example starts cycling above pin 7 after the hour is up, setting pins to low and un-setting the pinMode(switchPin,HIGH), breaking the internal pullup, and making the button non-functional.

The code should either reset the hourglass automatically, or check that it is in the 2-7 range before setting the pin to low.

Maybe change the test to:

    if (currentTime - previousTime > interval && led <= 7 ) {

Sped-up (6 seconds/step) Example:

https://wokwi.com/projects/422992836439284737

drf5n, 2025-02-16:
Here are a couple threads where the issue confused users:

https://forum.arduino.cc/t/digital-hourglass-logic/357157/4

https://forum.arduino.cc/t/arduino-starter-kit-example-project-8-digital-hourglass/358865 

And a discussion at:

https://forum.arduino.cc/t/builtin-example-10-p08-digitalhourglass-starter-kit-project-out-of-bounds-error/1354412

per1234, 2025-02-16:
Thanks for your report @drf5n. The sketches under the `10.StarterKit_BasicKit` folder are provided to accompany the **Arduino Projects Book** of the [**Arduino Starter Kit**](https://store.arduino.cc/products/arduino-starter-kit-multi-language). The **Arduino Projects Book** explains the code of these sketches and gives instructions for their usage.

The "**p08_DigitalHourglass**" sketch is unique in that it is intentionally left incomplete. The **Arduino Projects Book** instructs the user to add their own code to indicate the completion of the timer (which should also halt the timer code). The `if (led == 7)` block is annotated:

> Add one more if statement in the program to check if the LED on
pin 7 is turned on. Don’t do anything with this yet. You’ll decide
what happens at the end of the hour later.

And the tutorial instructions include:

> When the clock reaches one hour and all six lights are on, they just stay on. Can
you think of a way to get your attention when the hour is up? Sound or flashing
the lights are both good indicators. The led variable can be checked to see if all
the lights are on, that’s a good place to check for grabbing someone’s attention.

<sub>

<details markdown="1">
  <summary><b>©</b></summary>
  <p markdown="block">

[Arduino Projects Book](https://store.arduino.cc/products/arduino-starter-kit-multi-language) by [Arduino SA](https://www.arduino.cc/) - [CC BY-NC-SA 3.0](https://creativecommons.org/licenses/by-nc-sa/3.0/)

  </p>
</details>

</sub>

Since the primary purpose of the sketch is to accompany the **Arduino Starter Kit**, the code can not be changed here as you propose. The only way to do that would be to first update the code in the **Arduino Projects Book** and publish a new edition with the changes, then pull those changes into this repository. If you would like to propose that change, I think it will be best done via the official contact form, since the **Arduino Projects Book** content is not hosted in a public GitHub repository:

https://www.arduino.cc/en/contact-us

We do have discretion in the comments in these sketches, which were added to the sketches in this repository in order to provide the information that is not available to those users who don't have an **Arduino Projects Book**. I saw that the incomplete nature of this sketch was not documented and so I added comments to make this clear: https://github.com/arduino/arduino-examples/pull/84. Those comments will mitigate the problem you reported here.

closed 2025-02-16 by per1234